nopnop2002/Arduino-LoRa-Ra01S #2: Missing license for RadioLib code

jgromes, 2022-04-10:
Hello!

I'm the maintainer of the RadioLib project (https://github.com/jgromes/RadioLib), an open-source radio driver library. It seems like you have used a not-insignificant amount of code from that project - for example most of the register definitions in the Ra01S.h header (in RadioLib found in https://github.com/jgromes/RadioLib/blob/master/src/modules/SX126x/SX126x.h), as well as most low-level SPI access commands and SX126x commands (in RadioLib found in https://github.com/jgromes/RadioLib/blob/master/src/modules/SX126x/SX126x.cpp).

RadioLib uses MIT license, which allows to freely reuse any portion of the software for any purpose, however, the original license has to be stated. Therefore, to comply with RadioLib license, I would like to ask you to add the original RadioLib license (found here: https://github.com/jgromes/RadioLib/blob/master/license.txt) to this project.

Thanks!

nopnop2002, 2022-04-10:
Hello

I ported from here.
https://github.com/tinytronix/SX126x

jgromes, 2022-04-10:
That repository is also using code from RadioLib - in compliance with RadioLib license, as the original text is listed here: https://github.com/tinytronix/SX126x/blob/master/RadioLibLicense.txt

I also raised a similar issue in that repository a couple of years ago: https://github.com/tinytronix/SX126x/issues/2

It doesn't really matter where you got the code - if you're using licensed software, even if it is an open-source license, you still have to comply with it.

nopnop2002, 2022-04-10:
I added this.

https://github.com/nopnop2002/Arduino-LoRa-Ra01S/blob/main/RadioLibLicense.txt

jgromes, 2022-04-11:
Thanks, I think that should cover it then.

closed 2022-04-11 by jgromes