nopnop2002/Arduino-LoRa-Ra01S #5: Migration to Attiny

garudaonekh, 2022-11-12:
Hi,
I want to use this library with Attiny84 to build low cost LORA sensor. Is it hard to do? I have little knowledge on this low level programming.
As far as I know, attiny84 required softSPI or virtual SPI.

Thanks;

nopnop2002, 2022-11-12:
ATTINY has too little RAM to migrate.

garudaonekh, 2022-11-13:
Ok, it looks promising. After I removed all Serial.print from the your library, it's compiled successfully. Now, I have to figure out how to map the SPI pins.

```
Sketch uses 4610 bytes (56%) of program storage space. Maximum is 8192 bytes.
Global variables use 48 bytes (9%) of dynamic memory, leaving 464 bytes for local variables. Maximum is 512 bytes.
```

nopnop2002, 2022-11-13:
Good!!

This might work.
https://github.com/JChristensen/tinySPI
https://github.com/nickgammon/bitBangedSPI

garudaonekh, 2022-11-14:
Actually, AttinyCore already include its own virtual SPI handling so I don't need to use tinySPI anymore, I simply use SPI library of Arduino. However, I can't get it to work on attiny84, after several tries my Attiny84 are gone, I am waiting for new ones.
In the mean time, I try to make it work on Atmega328 first because it's easy to debug than attiny84 but I faced some issue, which I raise on another issue.

garudaonekh, 2022-11-29:
Hi, I decided to use Atmega16 instead because Attiny84 is already double the price of Atmega16a.
And I managed to make it works with atmega16a.

Thanks;

closed 2022-11-29 by garudaonekh