nopnop2002/Arduino-LoRa-Ra01S #6: Running on Arduino UNO

garudaonekh, 2022-11-14:
Hi,
When I compile it for Arduino UNO, 
I got this error

```
21:35:24.546 -> begin
21:35:24.546 -> debugPrint=1
21:35:24.546 -> SX126x_SPI_SELECT=10
21:35:24.546 -> SX126x_RESET=9
21:35:24.546 -> SX126x_BUSY=2
21:35:24.546 -> SX126x_TXEN=-1
21:35:24.546 -> SX126x_RXEN=-1
21:35:24.581 -> Reset
21:35:24.581 -> ReadRegister:  REG=0x740  DataIn: 0 84 
21:35:24.581 -> syncWord=0x84
21:35:24.581 -> SX126x error, maybe no SPI connection
```
0x84 is SX126X_CMD_SET_SLEEP.

I use the following PIN configs:
```
SX126x  lora(10,               //Port-Pin Output: SPI select
             0xff,               //Port-Pin Output: Reset 
             2               //Port-Pin Input:  Busy
             );
```

nopnop2002, 2022-11-14:
The initial value of syncWord for this module is 0x1424.

>21:35:24.581 -> syncWord=0x84

It indicates that the syncWord cannot be read correctly.

garudaonekh, 2022-11-15:
Thanks, I don't use RESET Pin to save pin and it works on ESP32 testinng. Is this fine because I only want to transmit data from sensor only.

nopnop2002, 2022-11-15:
If you don't use RESET pin, Do not connect wires.

This is not support

>0xff, //Port-Pin Output: Reset

garudaonekh, 2022-11-29:
Thanks
I use Atmega16a instead and it works. it's way much cheaper than atmega328.

closed 2022-11-29 by garudaonekh