nopnop2002/Arduino-LoRa-Ra01S #7: standby/sleep to reduce power

garudaonekh, 2022-11-29:
Hi,
I am designing remote control using Atmega16 and LLCC68(Ra-01SC)
My flow is when a user press a button, it will send a message and wait for response from the receiver and then it'll go to deep sleep. it'll run on 9v battery thus power consumption is very important. Can you advice me on the step in your library code?

I see you have standby function and there're several sleep mode constant which doesn't seem to related with each other.

Without doing anything(idle), this Ra-01sc consumes around 2ma which is very large.

Thanks;

nopnop2002, 2022-11-29:
LLCC68 has sleep and standby modes.

To use them, you need to read the datasheet and understand the register specifications.

Once you know the register specification, call the corresponding function.

garudaonekh, 2022-12-01:
[image]

I think setSleep function is not implemented in your library.

```
//SX126X_CMD_SET_SLEEP
#define SX126X_SLEEP_START_COLD                       0b00000000  //  2     2     sleep mode: cold start, configuration is lost (default)
#define SX126X_SLEEP_START_WARM                       0b00000100  //  2     2                 warm start, configuration is retained
#define SX126X_SLEEP_RTC_OFF                          0b00000000  //  0     0     wake on RTC timeout: disabled
#define SX126X_SLEEP_RTC_ON                           0b00000001  //  0     0                          enabled

```
I think the command are there in your header but function is not implemented

nopnop2002, 2022-12-01:
Added SetSleep function.

Please try the latest.

garudaonekh, 2022-12-01:
Here's my version of the functions:

```
void SX126x::sleep(bool retainConfig){
  SetStandby(SX126X_STANDBY_RC);
   uint8_t sleepMode = SX126X_SLEEP_START_WARM | SX126X_SLEEP_RTC_OFF;
  if(!retainConfig) {
    sleepMode = SX126X_SLEEP_START_COLD | SX126X_SLEEP_RTC_OFF;
  }
   WriteCommand(SX126X_CMD_SET_SLEEP, &sleepMode, 1);

  // wait for SX126x to safely enter sleep mode
  delay(1);

  return(0);
  
}
```

nopnop2002, 2022-12-01:
I haven't tried this code, but does it sleep correctly?

If it goes to Sleep state correctly, I will adopt your code.

garudaonekh, 2022-12-01:
> I haven't tried this code, but does it sleep correctly?
> 
> If it goes to Sleep state correctly, I will adopt your code.
> 
> Since it is a void function, return(0) is not necessary.

I will measure the current and feedback to you.

nopnop2002, 2022-12-01:
>I will measure the current and feedback to you.

Thnak you.

closed 2022-12-13 by garudaonekh