nopnop2002/Arduino-LoRa-Ra01S #8: Communicate with RadioLib

garudaonekh, 2022-12-06:
Hi,
I am unable to communicate between your lib and RadioLib. Here's both config. I recheck several times and it should be correct. Both side work normally with their own respective receiver/sender
```
#define RF_FREQUENCY                                433000000 // Hz  center frequency
#define TX_OUTPUT_POWER                             22        // dBm tx output power
#define LORA_BANDWIDTH                              4         // bandwidth
                                                              // 2: 31.25Khz
                                                              // 3: 62.5Khz
                                                              // 4: 125Khz
                                                              // 5: 250KHZ
                                                              // 6: 500Khz 
#define LORA_SPREADING_FACTOR                       9         // spreading factor [SF5..SF12]
#define LORA_CODINGRATE                             1         // [1: 4/5,
                                                              //  2: 4/6,
                                                              //  3: 4/7,
                                                              //  4: 4/8]

#define LORA_PREAMBLE_LENGTH                        8         // Same for Tx and Rx
#define LORA_PAYLOADLENGTH                          0         // 0: Variable length packet (explicit header)
```
Radiolib config:

```
#define frequency 433.0
#define bw 125.0
#define sf 9
#define cr 5
#define syncword 0x12
#define dbm 22
#define preamble 8
 int state = radio.begin(frequency,bw,sf,cr,syncword,dbm,preamble, 0, false);
```

nopnop2002, 2022-12-06:
syncword is 0x1424 for private networks.

garudaonekh, 2022-12-06:
> syncword is 0x1424 for private networks.

still can't receive. Radiolib is receiver.

nopnop2002, 2022-12-06:
Please try with this setting
```
  // Set frequency: 866Mhz
  // Set bandwidth(BW): 125Khz
  // Set Spreading Factor(SF): 7
  // Set Error Cording Rate(CR): 4/5
  // Set SyncWord: 0x1424(Private Network)
  // Set Power: 10dBm
  // Set Preamble Length: 8
  // Configure the radio to NOT use a TCXO controlled by DIO3
  // Set regulator mode: DC-DC
  int state = radio.begin(866.0, 125.0, 7, 5, RADIOLIB_SX126X_SYNC_WORD_PRIVATE, 10, 8, 0.0, false);
```