nopnop2002/Arduino-LoRa-Ra01S #11: SetTx Illegal Status Arduino Uno

johnpiker, 2023-07-10:
Hi, I am trying to implement your code on this module [()](https://www.waveshare.com/sx1262-lorawan-hat.htm) which has sx1262 chip. When I upload the code, I'm getting the following error. 
Can you please help. 

```
begin
debugPrint=0
SX126x_SPI_SELECT=5
SX126x_RESET=6
SX126x_BUSY=7
SX126x_TXEN=-1
SX126x_RXEN=-1
Reset
syncWord=0x1424
SX126x installed
tcxoVoltage=0.00
useRegulatorLDO=0
SetTx Illegal Status
```

nopnop2002, 2023-07-10:
LoRa modules with SX1262/1268 have several options.   

- Using TCXO(Temperature-Compensated Crystal Oscillator)   
SX1262/1268 can use the TCXO.   
If the TCXO is used, the XTB pin is not connected.   
However, the 6th pin (DIO3) of the SX1262/1268 can be used to power the TCXO.   
Explanation for TXCO and antenna control is [here](https://github.com/beegee-tokyo/SX126x-Arduino).   
Ra-01S / Ra-01SH does not use TCXO.   

- Power supply modes   
SX1262/1268 has two power supply modes.   
One is that only LDO used in all modes.   
Another is that DC_DC+LDO used for STBY_XOSC,FS, RX and TX modes.   
Explanation for LDO and DCDC selection is [here](https://github.com/beegee-tokyo/SX126x-Arduino).   
Ra-01S / Ra-01SH use only LDO in all modes.

- RF-Switching   
In general, use DIO2 to switch the RF-Switch.   
However, some tranceiver use an external gpio to switch the RF-Switch.   
Ra-01S / Ra-01SH use the SC70-6 integrated load switch to switch between RFO and RFI.   
Ra-01S / Ra-01SH use DIO2 to control this.   
DIO2 = 1, CTRL = 0, RFC for SC70-6 goes to RF1 -->  Switch to Tx Mode.   
DIO2 = 0, CTRL = 1, RFC for SC70-6 gose to RF2 -->  Switch to Rx Mode.   

You need to look at the schematic to set these options properly, but it's very esoteric.

ArshaanTech, 2023-10-11:
Hello!
I am getting the same error... Can you share your solution??

nopnop2002, 2023-10-11:
If this change doesn't work, I don't know how to solve it.

```
/*
  int16_t ret = lora.begin(RF_FREQUENCY,              //frequency in Hz
                           TX_OUTPUT_POWER);          //tx power in dBm
*/

  int16_t ret = lora.begin(RF_FREQUENCY,              //frequency in Hz
                           TX_OUTPUT_POWER,           //tx power in dBm
                           3.3,                       //use TCXO
                           true);                     //use TCXO
```

ArshaanTech, 2023-10-12:
It still gives following error:

begin
debugPrint=0
SX126x_SPI_SELECT=5
SX126x_RESET=6
SX126x_BUSY=7
SX126x_TXEN=-1
SX126x_RXEN=-1
Reset
syncWord=0x1424
SX126x installed
tcxoVoltage=3.30
useRegulatorLDO=1
SPI Transaction error:10
SPI Transaction error:10
SetRx Illegal Status

Btw I have not connected any of the TXEN or RXEN pin of LoRa. The DIO pins are also disconnected.

nopnop2002, 2023-10-12:
Contacting the LoRa module vendor may provide a solution.

ArshaanTech, 2023-10-13:
Still not working... I instead used the RadioLib library which is working fine...

ArshaanTech, 2023-10-13:
Still not working...