nopnop2002/Arduino-LoRa-Ra01S #15: Unstable Behavior in EBYTE E220-400M22S Setup with ATTINY804

krishank652, 2024-02-06:
I am seeking assistance in the library GitHub to address an issue I'm encountering.

My challenge involves establishing communication between the Ebyte LoRa E220-400M22S and the MCU ATTINY804 IC. Below is the code snippet I'm utilizing:

```cpp

#include <Ra01S.h>

#define RF_FREQUENCY                                433000000 // Hz  center frequency
//#define RF_FREQUENCY                                866000000 // Hz  center frequency
//#define RF_FREQUENCY                                915000000 // Hz  center frequency
#define TX_OUTPUT_POWER                             22        // dBm tx output power
#define LORA_BANDWIDTH                              4         // bandwidth
                                                              // 2: 31.25Khz
                                                              // 3: 62.5Khz
                                                              // 4: 125Khz
                                                              // 5: 250KHZ
                                                              // 6: 500Khz                                                               
#define LORA_SPREADING_FACTOR                       7         // spreading factor [SF5..SF12]
#define LORA_CODINGRATE                             1         // [1: 4/5,
                                                              //  2: 4/6,
                                                              //  3: 4/7,
                                                              //  4: 4/8]

#define LORA_PREAMBLE_LENGTH                        8         // Same for Tx and Rx
#define LORA_PAYLOADLENGTH                          0         // 0: Variable length packet (explicit header)
                                                              // 1..255  Fixed length packet (implicit header)
#if 1

SX126x  lora(0,               //Port-Pin Output: SPI select
             1,               //Port-Pin Output: Reset 
             2                //Port-Pin Input:  Busy
             );
#endif // ATmega328/2560

#if 0

SX126x  lora(0,                 //Port-Pin Output: SPI select
             1,                 //Port-Pin Output: Reset 
             2                 //Port-Pin Input:  Busy
             );
#endif // ESP8266

void setup() 
{
  delay(1000);
  Serial.begin(9600);

  lora.DebugPrint(true);

 /*
  int16_t ret = lora.begin(RF_FREQUENCY,              //frequency in Hz
                           TX_OUTPUT_POWER);          //tx power in dBm
*/

// UPDATED BELOW CODE AS PER INSTRUCTION ON LIBRARY PAGE

  int16_t ret = lora.begin(RF_FREQUENCY,              //frequency in Hz
                           TX_OUTPUT_POWER,           //tx power in dBm
                           3.3,                       //use TCXO
                           true);                     //use TCXO

                           
  if (ret != ERR_NONE) while(1) {delay(1);}

  lora.LoRaConfig(LORA_SPREADING_FACTOR, 
                  LORA_BANDWIDTH, 
                  LORA_CODINGRATE, 
                  LORA_PREAMBLE_LENGTH, 
                  LORA_PAYLOADLENGTH, 
                  true,               //crcOn  
                  false);             //invertIrq

}

void loop() 
{
  uint8_t txData[255];
  sprintf((char *)txData, "Hello World %lu", millis());
  uint8_t len = strlen((char *)txData);

  // Wait for transmission to complete
  if (lora.Send(txData, len, SX126x_TXMODE_SYNC)) {
    //Serial.println("Send success");
  } else {
    Serial.println("Send fail");
  }

  // Do not wait for the transmission to be completed
  //lora.Send(txData, len, SX126x_TXMODE_ASYNC );

  delay(1000);
}
```

I used above code from the library Example: [example/Ra01S-TX/Ra01S-TX.ino](https://github.com/nopnop2002/Arduino-LoRa-Ra01S/blob/main/example/Ra01S-TX/Ra01S-TX.ino) and changes are done by me :

1. Uncomment the 433MHz Line and Comment the 915MHz line.
```
#define RF_FREQUENCY                                433000000 // Hz  center frequency
//#define RF_FREQUENCY                                866000000 // Hz  center frequency
//#define RF_FREQUENCY                                915000000 // Hz  center frequency
```
2. Updated below as per my connections I am using ATTINY804 connections details are in the below.
```
#if 1

SX126x  lora(0,               //Port-Pin Output: SPI select
             1,               //Port-Pin Output: Reset 
             2                //Port-Pin Input:  Busy
             );
#endif // ATmega328/2560

#if 0

SX126x  lora(0,                 //Port-Pin Output: SPI select
             1,                 //Port-Pin Output: Reset 
             2                 //Port-Pin Input:  Busy
             );
#endif // ESP8266
```
3. Below Line for serial monitor
  `Serial.begin(9600);`
4.  Uncomment below line for debugging.
`  lora.DebugPrint(true);`
5.  As per instruction given on the library page I am using Ebyte module So I updated below code:
 ```
/*
  int16_t ret = lora.begin(RF_FREQUENCY,              //frequency in Hz
                           TX_OUTPUT_POWER);          //tx power in dBm
*/

// UPDATED BELOW CODE AS PER INSTRUCTION ON LIBRARY PAGE

  int16_t ret = lora.begin(RF_FREQUENCY,              //frequency in Hz
                           TX_OUTPUT_POWER,           //tx power in dBm
                           3.3,                       //use TCXO
                           true);                     //use TCXO
```

During the initial startup, the Serial Monitor output is as follows:

```plaintext
21:25:27.203 -> begin
21:25:27.203 -> debugPrint=1
21:25:27.203 -> SX126x_SPI_SELECT=0
21:25:27.250 -> SX126x_RESET=1
21:25:27.250 -> SX126
21:25:27.250 -> 
21:25:27.250 -> 
21:25:28.187 -> ut: 7-->A2 0-->A2 1-->A2 40-->A2 
21:25:28.234 -> WriteCommand:  CMD=0x89  DatD
21:25:28.282 -> 
21:25:28.282 -> 
21:25:29.219 -> =0x8  DataOut: 3-->A2 FF-->A2 0-->A2 0-->A2 0-->A2 0-->A2 0-->A
21:25:29.266 -> 
21:25:29.266 ->
```

However, subsequent startups after a power restore yield different results, suggesting an inconsistency issue:

```plaintext
21:26:56.403 -> ⸮
21:26:56.403 -> 
21:26:56.403 -> 
21:26:57.402 -> begin
21:26:57.402 -> debugPrint=1
21:26:57.449 -> SX126x_SPI_SELECT=0
21:26:57.449 -> SX126x_RESET=1
21:26:57.496 -> SX126
21:26:57.496 -> 
21:26:57.496 -> 
21:26:58.402 -> 0x1424
21:26:58.449 -> SX126x installed
21:26:58.449 -> WriteCommand:  CMD=0x80  DataOut: 0--
21:26:58.496 -> 
21:26:58.496 ->
```

These inconsistencies occur with subsequent startups, implying a problem with the setup. 

The connections between ATTINY804 and E220-400M22S are as follows:

- SCK <> SCK
- MOSI <> MOSI
- MISO <> MISO
- 0 <> NSS
- 1 <> NRST
- 2 <> BUSY

For power supply, I'm using the AMS1117 3.3V Power Supply Module, drawing input from Arduino 5V and GND, then supplying it to the circuit via the AMS1117 3.3V Power supply.

For ATTINY804 programming, I'm utilizing the [megaTinyCore Library](https://github.com/SpenceKonde/megaTinyCore/tree/master) and referring to the [ATTINY804 pinout ](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ATtiny_x04.md)details.

I'm reaching out to the library developer and the community for assistance in resolving this issue. Any guidance or suggestions would be greatly appreciated. Thank you in advance.

nopnop2002, 2024-02-06:
It looks like Serial.print is not working correctly.

krishank652, 2024-02-07:
> It looks like Serial.print is not working correctly.

Thanks for the reply.
For this what I need to do? Is I need to change baud rate or anything other things?

nopnop2002, 2024-02-07:
I've never used ATTINY804, so I don't know.

nopnop2002, 2024-02-12:
Thank you for the report.

There were missing changes when using the EBYTE module.

If you use the EBYTE module, the following changes are required.

```
/*
SX126x  lora(5,               //Port-Pin Output: SPI select
             6,               //Port-Pin Output: Reset 
             7                //Port-Pin Input:  Busy
             );

int16_t ret = lora.begin(RF_FREQUENCY,              //frequency in Hz
                         TX_OUTPUT_POWER);          //tx power in dBm
*/

SX126x  lora(5,               //Port-Pin Output: SPI select
             6,               //Port-Pin Output: Reset 
             7                //Port-Pin Input:  Busy
             8                //Port-Pin Output: TXEN
             9                //Port-Pin Output: RXEN
             );

int16_t ret = lora.begin(RF_FREQUENCY,              //frequency in Hz
                         TX_OUTPUT_POWER,           //tx power in dBm
                         3.3,                       //use TCXO
                         true);                     //use TCXO
```

Two additional wires are required.   
|EBYTE||UNO|MEGA|ESP8266|
|:-:|:-:|:-:|:-:|:-:|
|TXEN|--|8(*3)|8(*3)|D4|
|RXEN|--|9(*3)|9(*3)|D5|

(*3)   
SX126x is not 5V tolerant.   
You need level shift from 5V to 3.3V.

krishank652, 2024-02-24:
Thanks for the update....

I have again taken trial but still unable to work. Can You please see the again. this help will make me happy😄

I updated following things:
1. Circuit Updated.
2. Library Updated.
3. Used New Tx and Rx Code.

New Circuit
[image]

Transmitter Code:

```
#include <Ra01S.h>

#define RF_FREQUENCY                                433000000 // Hz  center frequency
//#define RF_FREQUENCY                                866000000 // Hz  center frequency
//#define RF_FREQUENCY                                915000000 // Hz  center frequency
#define TX_OUTPUT_POWER                             22        // dBm tx output power
#define LORA_BANDWIDTH                              4         // bandwidth
                                                              // 2: 31.25Khz
                                                              // 3: 62.5Khz
                                                              // 4: 125Khz
                                                              // 5: 250KHZ
                                                              // 6: 500Khz                                                               
#define LORA_SPREADING_FACTOR                       7         // spreading factor [SF5..SF12]
#define LORA_CODINGRATE                             1         // [1: 4/5,
                                                              //  2: 4/6,
                                                              //  3: 4/7,
                                                              //  4: 4/8]

#define LORA_PREAMBLE_LENGTH                        8         // Same for Tx and Rx
#define LORA_PAYLOADLENGTH                          0         // 0: Variable length packet (explicit header)
                                                              // 1..255  Fixed length packet (implicit header)

#define USE_EBYTE

#if 1
/*
 * for ATmega328/2560
 * VCC    3V3/3V3
 * GND    GND/GND
 * SCK    13/52
 * MISO   12/50
 * MOSI   11/51
 * NSS     5/5
 * RST     6/6
 * BUSY    7/7
 * TXEN    8/8 for EBYTE
 * RXEN    9/9 for EBYTE
 */

#ifdef USE_EBYTE
SX126x  lora(10,               //Port-Pin Output: SPI select
             6,               //Port-Pin Output: Reset 
             7,               //Port-Pin Input:  Busy
             2,               //Port-Pin Output: TXEN
             3                //Port-Pin Output: RXEN
             );

#else
SX126x  lora(5,               //Port-Pin Output: SPI select
             6,               //Port-Pin Output: Reset 
             7                //Port-Pin Input:  Busy
             );
#endif // USE_EBYTE

#endif // ATmega328/2560
[... 336 lines cut ...]
21:44:29.944 -> WriteCommand:  CMD=0x8C  DataOut: 0-->A2 8-->A2 0-->A2 FF-->A2 1-->A2 0-->A2 
21:44:29.944 -> WriteCommand:  CMD=0x8  DataOut: 3-->A2 FF-->A2 0-->A2 0-->A2 0-->A2 0-->A2 0-->A2 0-->A2 
21:44:29.944 -> ----- SetRx timeout=16777215
21:44:29.944 -> WriteCommand:  CMD=0x80  DataOut: 0-->A2 
21:44:29.944 -> SetRxEnable:SX126x_TXEN=2 SX126x_RXEN=3
21:44:29.991 -> WriteCommand:  CMD=0x82  DataOut: FF-->A2 FF-->A2 FF-->A2 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> ReadCommand:   CMD=0xC0  DataIn: 2A 
21:44:29.991 -> SetRx Illegal Status
```

Please see again and please help me to build my device. thanks.

One thing is that: I see this line: `SX126x installed` it means my LoRa device is working?

Please help me.

krishank652, 2024-02-25:
Dear,
Can you please solve the problem. I have E220-400M22S module. Can you please solve the error with the library. 
Thanks

nopnop2002, 2024-02-25:
I also have an E220-400M22S, but I don't know the solution.